INTI-CMNB/KiBot #810: [Query] When can KiBot run without a schematic file?

Wol, 2025-05-01:
I'm trying to get KiBot (via kici) to generate fabrication files and images for a kicad_pcb file where I don't have a kicad_sch file to along with it (it's a panellised version of a normal kicad project, so doesn't work with a schematic)

In another thread, I saw:

> KiBot can work without an schematic, but only in particular cases. 
  
 _Originally posted by @set-soft in [#661](https://github.com/INTI-CMNB/KiBot/issues/661#issuecomment-2352706394)_

Is it possible to get clarification on which particular cases these would be? 

The particular point I'm currently stuck on is when it's creating the JLCPCB outputs. `- 'Gerbers compatible with JLCPCB' (_JLCPCB_gerbers) [gerber]` `- 'Drill files compatible with JLCPCB' (_JLCPCB_drill) [excellon]` seem to complete okay, but then it fails just after it prints `- 'Pick and place file, JLCPCB style' (_JLCPCB_position) [position]`. I _think_ that this is because it's pulling in `VariantOptions` in `out_position.py` which in turn tries loading the schematic file. 

Is there a way to stop it trying to load the schematic, or to skip variants altogether?

set-soft, 2025-05-01:
Variants needs an schematic.
For a panel just use the original schematic.

oliv3r, 2025-05-13:
@Wol what prevents you from having a dump schematic with nothing in it? Maybe just a Text Label or copyright notice or ... I don't even think you NEED anything in it. And then just do F8?

Wol, 2025-05-13:
@oliv3r Nothing is preventing me from having a blank schematic file, I just wanted to make sure that there wasn't a secret option somewhere that would make it work without it, and also to clarify that it's not going to start pulling in odd data by having a mismatched schematic!

Shall give that a try then. Thanks both for your answers!

oliv3r, 2025-05-13:
Fair enough :) I noticed a similar thing, where I had a pcb file from a hierarchical schematic. Things stopped working then too, and needed some help (kici/world ;)

closed 2025-05-20 by set-soft