INTI-CMNB/KiBot #812: [BUG] "Private" text fields on symbols throw error

Wol, 2025-05-06:
When using a symbol which has a "private" property/field in it, I get the following error:

`ERROR:field name is not a string `Symbol('private')` (kibot.gs - gs.py:897)`

The "private" basic property when set in KiCad:

[image]

makes the .sch file look like:

`(property private "Datasheet Link" "datasheet_url.pdf"`

When KiBot tries to parse this, field.name is loaded as "Symbol(private)" which then `_check_str` checks for and throws the option. 

Removing the "private" tag allows it to build correctly. 

This is within the docker image: sha256:6e145be9e938b4a511877e2faf93208066827ec179c99bc6e8bca8357bef3b6d for ghcr.io/inti-cmnb/kicad9_auto_full:dev with digest ghcr.io/inti-cmnb/kicad9_auto_full@sha256:8c01e863aaf0e36f5c2b5ca319d2df45e0bba4b8aa1ad0939618b5617c39d765 

`$ kibot --version
KiBot 1.8.5 - Copyright 2018-2025, Salvador E. Tropea/INTI/John Beard - License: AGPL v3`

set-soft, 2025-05-06:
I'm changing it to "Feature" because this is not documented.

closed 2025-05-06 by set-soft

Wol, 2025-05-07:
Thanks for the fast fix!