INTI-CMNB/KiBot #813: [BUG] DRC error only with KiBot (not in KiCad) for pads with offset holes

fleborne, 2025-05-06:
**Describe the bug**

When a PCB has pads with a non-null "offset shape from hole" , the KiBot DRC reports errors, despite the KiCad DRC reporting no error.

The errors are:
```
ERROR:2 DRC errors detected (kibot.pre_drc - pre_drc.py:151)
ERROR:(annular_width) Annular width (board setup constraints min annular width 0.0500 mm; actual 0.0000 mm)
    @(102.35 mm, 87.8 mm): PTH pad MP [GND] of J2
    Check: Violations (kibot.pre_drc - pre_drc.py:166)
ERROR:(annular_width) Annular width (board setup constraints min annular width 0.0500 mm; actual 0.0000 mm)
    @(102.35 mm, 95.0 mm): PTH pad MP [GND] of J2
    Check: Violations (kibot.pre_drc - pre_drc.py:166)
ERROR:DRC errors: 2 (kibot.gs - gs.py:899)
```

**To Reproduce**

KiCad Version: 8.0.9-8.0.9-0~ubuntu22.04.1, release build
KiBot Version: 1.8.4

Create a footprint with at least one PTH pad, and set it's "offset shape from hole" values to something non-null.
Run KiCad's DRC from the user interface, and KiBot's DRC with the preflight rule.

**Expected behavior**

No error reported by KiBot's DRC when KiCad's DRC reports none.

**Screenshots**

[image]

[image]

**Environment (please complete the following information):**

I run KiBot using the `ghcr.io/inti-cmnb/kicad8_auto_full:latest` Docker image.

set-soft, 2025-05-06:
Hi @fleborne !
Are you using `erc` or `run_erc`?
Can you create a small example?
Looks like a bug in KiCad CLI.

fleborne, 2025-05-07:
Hi @set-soft ,

I have the same behavior with both `erc` and `run_erc`.
There is a minimal example that demonstrates this issue in the attachment.

[issue813_offset_hole.zip](https://github.com/user-attachments/files/20078257/issue813_offset_hole.zip)

set-soft, 2025-05-07:
Hi @fleborne !
I can confirm this is a bug in KiCad 8, i.e.:

```shell
$ kibot
Using config file: issue813_offset_hole.kibot.yaml
Using SCH file: issue813_offset_hole.kicad_sch
- Running the DRC
WARNING:(W145) 1 DRC warnings detected (kibot.pre_drc - pre_drc.py:150)
WARNING:(W145) (lib_footprint_issues) The current configuration does not include the library 'Connectors'.
    @(134.95 mm, 88.5 mm): Footprint J1
    Check: Violations (kibot.pre_drc - pre_drc.py:150)
ERROR:2 DRC errors detected (kibot.pre_drc - pre_drc.py:151)
ERROR:(annular_width) Annular width (board setup constraints min annular width 0.1000 mm; actual 0.0000 mm)
    @(134.95 mm, 84.9 mm): PTH pad MP [GND] of J1
    Check: Violations (kibot.pre_drc - pre_drc.py:166)
ERROR:(annular_width) Annular width (board setup constraints min annular width 0.1000 mm; actual 0.0000 mm)
    @(134.95 mm, 92.1 mm): PTH pad MP [GND] of J1
    Check: Violations (kibot.pre_drc - pre_drc.py:166)
ERROR:DRC errors: 2 (kibot.gs - gs.py:899)

$ kicad-cli pcb drc *pcb
Loading board
Running DRC...
Found 3 violations
Found 0 unconnected items
Saved DRC Report to issue813_offset_hole.rpt

$ cat issue813_offset_hole.rpt
** Drc report for issue813_offset_hole.kicad_pcb **
** Created on 2025-05-07T09:38:41+0000 **

** Found 3 DRC violations **
[annular_width]: Annular width (board setup constraints min annular width 0.1000 mm; actual 0.0000 mm)
    Rule: board setup constraints; error
    @(134.9500 mm, 84.9000 mm): PTH pad MP [GND] of J1
[annular_width]: Annular width (board setup constraints min annular width 0.1000 mm; actual 0.0000 mm)
    Rule: board setup constraints; error
    @(134.9500 mm, 92.1000 mm): PTH pad MP [GND] of J1
[lib_footprint_issues]: The current configuration does not include the library 'Connectors'.
    Local override; warning
    @(134.9500 mm, 88.5000 mm): Footprint J1

** Found 0 unconnected pads **

** Found 0 Footprint errors **

** End of Report **
```

And this error isn't found in KiCad 9:

```
$ kibot 
Using config file: issue813_offset_hole.kibot.yaml
Using SCH file: issue813_offset_hole.kicad_sch
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:375)
- Running the DRC
WARNING:(W145) 1 DRC warnings detected (kibot.pre_drc - pre_drc.py:150)
WARNING:(W145) (lib_footprint_issues) The current configuration does not include the footprint library 'Connectors'.
    @(134.95 mm, 88.5 mm): Footprint J1
    Check: Violations (kibot.pre_drc - pre_drc.py:150)
- 'Interactive HTML BOM' (ibom) [ibom]
Found 3 unique warning/s (3 total)
```

As KiCad team is no longer maintaining KiCad 8 you should migrate to KiCad 9. Of course you can just filter them using:

```yaml
kibot:
  version: 1

preflight:
  check_zone_fills: true
  drc:
    enabled: true
    filters:
      - error: 'annular_width'
        filter: 'Ignore bogus annular width error'
        regex: 'PTH pad (.*) of J1'
  erc:
    enabled: false
```

closed 2025-05-07 by set-soft