INTI-CMNB/KiBot #814: [FEATURE] Support Net Class directives on rule areas in KiCad v9

Wol, 2025-05-07:
Trying to put a directive label on a rule area is throwing an ERC error in KiBot, but not in KiCad

[image]

```
- Running the ERC
ERROR:1 ERC errors detected (kibot.pre_erc - pre_erc.py:147)
ERROR:(label_dangling) Label not connected to anything
    @(1.4859 mm, 0.762 mm): Directive Label [Net Class SMALL_TRACE]
    Sheet: / (kibot.pre_erc - pre_erc.py:162)
ERROR:ERC errors: 1 (kibot.gs - gs.py:899)
```

Is it possible to include support for this to fix the label error?

Currently running KiBot 1.8.5 via kicad9_auto_full:dev docker container. Not sure which exact commit this is running from however. (Is there a way to find this in `kibot -V` or similar)

```
dev: Pulling from inti-cmnb/kicad9_auto_full
Digest: sha256:ff00172a92f90a239802c8871994fc68b88e36b7254a68d51c158e7a521a0064
Status: Image is up to date for ghcr.io/inti-cmnb/kicad9_auto_full:dev
```

set-soft, 2025-05-07:
Can you post this small example?

set-soft, 2025-05-07:
Hi @Wol !
Is a KiCad bug, already reported here: https://gitlab.com/kicad/code/kicad/-/issues/20528

set-soft, 2025-05-07:
You can just filter it until the issue is fixed.

```yaml
# Example KiBot config file
kibot:
  version: 1

preflight:
  erc:
     filters:
       - filter: 'Ignore bogus dangling label'
         error: label_dangling
         regex: 'SMALL_TRACE'
```

Wol, 2025-05-07:
Ah thanks for the link :) I'll keep an eye on it!