INTI-CMNB/KiBot #816: [HOW] Use blender export

marcoscas, 2025-05-13:
**What do you want to achieve?**
Firstly, I am new to KiBot. I have been trying to create a Blender export for several hours and am encountering a problem: the component pads are consistently positioned outside of the PCB. When I generate the pcb3D file directly from KiCad, it is correct; however, the file created with KiBot is always incorrect. I have tried numerous options but have been unable to generate a correct file. It appears to be a configuration issue, and I would like to ask if anyone has experienced the same problem.
One difference I noticed is that the KiCad output includes several files with the .toml extensión, which are readable with a text editor. However,, the KiBot output files have no extension and cannot be openend with a text editor.

**Do you have some PCB/Schematic to use as example?**
Attached is a sample project with the output generated from KiCad (test_3d_fromKicad) and from KiBot (test_3d_blender_export).

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
kibot:
  version: 1

outputs:
  - name: '3d_export'
    comment: "Exports the PCB in blender format"
    type: blender_export
    options:
      render_options:
        transparent_background: true
        samples: 10
        #resolution_x: 1920
        #resolution_y: 1080
      point_of_view:
        rotate_x: 30
        rotate_z: -20
        # view: bottom
      outputs:
        - type: blender
        - type: render
```

**Environment (please complete the following information):**
I am using Tuxedo OS (Ubuntu 24.04) with Kicad installed from PPA.

**Additional context**
This behavior occurs in all of my projects, which leads me to believe there is an issue with my installation.

[test_3d.zip](https://github.com/user-attachments/files/20184760/test_3d.zip)

set-soft, 2025-05-13:
Hi @marcoscas !
The problem seems to be in the `pcb2blender` version you are using. 
It looks like 2.17 can't read files created by 2.16.

Running your config using the 2.16 plug-in  I get:

[image]

Where the pads are inside the PCB.

But when using 2.17.0 things go wrong.

I'll downgrade the version installed in KiBot docker images and will investigate the problem.

set-soft, 2025-05-13:
You need importer v2.16
This means that you'll have to generate your pcb3d files using KiBot, which can generate 2.16 files for KiCad 9.
Lamentably v2.17 importer can read v2.16 exporter files, but has this issue with the pads outside the PCB.
Right now KiBot can't generate v2.17 files.

set-soft, 2025-05-13:
Looks like 2.17.2, released 8 minutes ago, fixes it. I'll check it

set-soft, 2025-05-13:
Ok, confirmed, use 2.17.2 version, it fixes the pads issue:
https://github.com/30350n/pcb2blender/issues/110

closed 2025-05-13 by set-soft

marcoscas, 2025-05-13:
Yes, I have checked. Thank you!!