INTI-CMNB/KiBot #818: [BUG] The 3D model selection mechanism ("complex") is not working with ghcr.io/inti-cmnb/kicad9_auto_full:1.8.4

mdeweerd, 2025-05-16:
**Describe the bug**
The 3D model selection mechanism ("complex") is not working with ghcr.io/inti-cmnb/kicad9_auto_full:1.8.4 .

**To Reproduce**
Generate outputs using example:
https://github.com/INTI-CMNB/KiBot/tree/master/docs/samples/3D_Model_LCD

All image outputs are the same while they should differ according to the variant.

set-soft, 2025-05-16:
Which version was the last working?

mdeweerd, 2025-05-16:
It looks like it has to do with the KiCAD version.

kicad6 works up to 1.8.4 included (dev not tested).
kicad7 does not work since 1.6.1 (minimum version tested).
kicad8 did not work either

Tested like this where runInKiAuto is a wrapper around the docker call, and where KVER sets the tag of the docker image.

```bash
for v in 1.6.1 1.6.2 1.6.3 1.6.4 1.6.5 1.7.0 1.8.1 1.8.2 1.8.3 1.8.4 dev ; do
  KVER=$v ./runInKiAuto kibot -d output_$v
  printf "\n$v " |& tee -a versions.log
  compare -metric MAE output_$v/lcd-3D_top_variant_left.png output_$v/lcd-3D_top_variant_top.png null: |&
 tee -a versions.log
done
echo ""
```

Output (the bigger the number, the bigger the difference - less than 25 is "equal" in the the tests, more than 3600 is different::
```log
# kicad6
1.8.4 3689.98 (0.0563054)
# kicad 7
1.6.1 22.1823 (0.000338481)
```

mdeweerd, 2025-05-17:
FP_TEXT was renamed to PCB_TEXT:
https://gitlab.com/kicad/code/kicad/-/commit/37906511f585fd635a0d8a233472c79406407dee

More specifically, "MTEXT" was changed to "PCB_TEXT"
https://gitlab.com/kicad/code/kicad/-/blame/master/pcbnew/pcb_text.h?ref_type=heads#L122

So kibot should test against MTEXT and PCB_TEXT.

mdeweerd, 2025-05-17:
I am closing this (fix), another fix #820 is related but fixes the handling of default.

closed 2025-05-17 by mdeweerd

set-soft, 2025-05-19:
> I am closing this (fix), another fix [#820](https://github.com/INTI-CMNB/KiBot/pull/820) is related but fixes the handling of default.

I added a review to #820, please make the change as small as possible (no extra rename)

commit 3ec9b8da (2025-05-17): Merge pull request #819 from mdeweerd/fix/818

fix #818 (3D models): update 3D variant aspect mechanism to check for PCB_TEXT