INTI-CMNB/KiBot #821: [BUG] KiBot does not handle 'kicad-embed://' files during VRML export

oliv3r, 2025-05-19:
**Describe the bug**
During pcb3d generation, we need to have the component that are embedded in KiCad via `kicad-embed://` somewhere available, as kibot later relies on it. Not having the embedded model available results in
https://gitlab.com/ci-includes/kici/-/jobs/10073606135
```
ERROR:Error from Blender run:
Traceback (most recent call last):
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 132, in execute
    if not isinstance(result := self.import_pcb3d(context, filepath), PCB3D):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 492, in import_pcb3d
    component = self.component_cache[url]
                ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: 'components/kicad_embedded_0CDAE257EC704CDB6EB68A1B683DD6DC.wrl'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 464, in <module>
    main()
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 438, in main
    bpy.ops.pcb2blender.import_pcb3d(**ops)
  File "/usr/bin/4.2/scripts/modules/bpy/ops.py", line 109, in __call__
    ret = _op_call(self.idname_py(), kw)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 132, in execute
    if not isinstance(result := self.import_pcb3d(context, filepath), PCB3D):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 492, in import_pcb3d
    component = self.component_cache[url]
                ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: 'components/kicad_embedded_0CDAE257EC704CDB6EB68A1B683DD6DC.wrl'
Location: /usr/bin/4.2/scripts/modules/bpy/ops.py:109
```

I'm not sure if we (should) expect kicad-cli to export this part, oth if it doesn't export it, our pcb3d file will be incomplete. So I'll log a bug report with kicad for the 'copy' function as well. Meanwhile, kibot should probably extend the `copy2` function recently added to extract and copy embedded models with a warning?

**To Reproduce**
% cd /tmp
% docker run \
    --env HOME='/tmp' \
    --interactive \
    --pull always \
    --rm \
    --tty \
    --user "$(id -u):$(id -g)" \
    --volume "$(pwd):/workdir" \
    --workdir "/workdir" \
    'ghcr.io/inti-cmnb/kicad9_auto_full:dev' \
    '/bin/bash'
$ git clone https://gitlab.com/ci-includes/kici.git
$ cd kici/world
$ git reset --hard e697f5b155
$ export CI_JOB_NAME='KiCi'
$ kibot --schematic 'world.kicad_sch' --defs-from-env --stop-on-warnings --plot-config '.kibot/kici_blender.kibot.yaml'

**Expected behavior**
No error, and extracted model so kibot can render in blender.

oliv3r, 2025-05-19:
Right, the first time kicad-cli runs, it puts the needed files in the cache, but doesn't copy it.

```
I have no name!@3b219845e296:/workdir/kici/world$ rm -rf /tmp/kibot_tmp/
I have no name!@3b219845e296:/workdir/kici/world$ rm -rf /tmp/.cache/
I have no name!@3b219845e296:/workdir/kici/world$ kicad-cli pcb export vrml -o /tmp/kibot_tmp/pcb.wrl -f --units m --models-relative --models-dir components --user-origin 0.0x0.0mm world.kicad_pcb
Successfully exported VRML to /tmp/kibot_tmp/pcb.wrl
I have no name!@3b219845e296:/workdir/kici/world$ find /tmp -iname '*.wrl'
/tmp/.cache/kicad/9.0/embed/kicad_embedded_0CDAE257EC704CDB6EB68A1B683DD6DC.wrl
/tmp/kibot_tmp/components/VSSOP-8_3x3mm_P0.65mm.wrl
/tmp/kibot_tmp/components/LED_0805_2012Metric.wrl
/tmp/kibot_tmp/components/CP_Elec_5x3.wrl
/tmp/kibot_tmp/components/R_0805_2012Metric.wrl
/tmp/kibot_tmp/pcb.wrl
I have no name!@3b219845e296:/workdir/kici/world$ kicad-cli pcb export vrml -o /tmp/kibot_tmp/pcb.wrl -f --units m --models-relative --models-dir components --user-origin 0.0x0.0mm world.kicad_pcb
Successfully exported VRML to /tmp/kibot_tmp/pcb.wrl
I have no name!@3b219845e296:/workdir/kici/world$ find /tmp -iname '*.wrl'
/tmp/.cache/kicad/9.0/embed/kicad_embedded_0CDAE257EC704CDB6EB68A1B683DD6DC.wrl
/tmp/kibot_tmp/components/VSSOP-8_3x3mm_P0.65mm.wrl
/tmp/kibot_tmp/components/LED_0805_2012Metric.wrl
/tmp/kibot_tmp/components/CP_Elec_5x3.wrl
/tmp/kibot_tmp/components/R_0805_2012Metric.wrl
/tmp/kibot_tmp/components/kicad_embedded_0CDAE257EC704CDB6EB68A1B683DD6DC.wrl
/tmp/kibot_tmp/pcb.wrl
```

But as you can see, running it a second time, it gets it from the cache, and then it does work. I wonder if https://gitlab.com/kicad/code/kicad/-/merge_requests/2241 basically 'fixes' this.

set-soft, 2025-05-19:
It looks like you want to add more features to a workaround for a KiCad bug. Why do you say this is a bug?

set-soft, 2025-05-19:
The above patch should also look for embedded models.

oliv3r, 2025-05-21:
@set-soft lol I don't want to fix every kicad bug here :) But I do want stuff to work, and as you often said yourself, KiCad is riddled with bugs you have to work around :(

When I initially opened this bug, I thought it was a KiBot issue to be fair; but alas, it's again a KiCad issue, though I do hope that 2241 fixes it, as it seems similar in behavior enough (caching issue). Whether this is a bug or a feature 'but after the first invocation it IS in the cache!' ... who knows if/when this will ever be 'fixed'. So a work-around is warranted, thank you for being so quick about it :D

One could also argue you already have a work-around, the work-around is incomplete :p I'll go test it and will confirm the fix works.

commit 66ca0b32 (2025-05-19): [VRML][Added] Extended the workaround for missing models to embedded ones

Closes #821
closed 2025-05-21 by set-soft