INTI-CMNB/KiBot #825: [FEATURE] Variant properties and/or pass kibot fields to "command" for set_text_variable.

mdeweerd, 2025-05-20:
I want to set "PRODUCT_CODE" appearing on the sheet according to the currently selected variant.
It's value would not be the variant name.

I tried using the "command" approach to set a text_variable, but it does not show the kibot variant configuration values in the environment and they are not passed along:

```yaml
preflight:
  set_text_variables:
    - name: CMD
      command: '(env ; echo "(%V)(%f)%f_%r_%d-%i%v.%x %f_%br_%bd_%g/") > test.log'
```

Maybe some environment variables could be set to make the configuration information available to "command".

An alternative approach could be to allow setting a varaible in the variant definition, with a default value set elsewhere.

Maybe I missed another method that would make this possible.

set-soft, 2025-05-20:
Did you try `expand_in_command`

```yaml
kibot:
  version: 1

variants:
  - name: MY_VARIANT
    type: kibom

preflight:
  set_text_variables:
    - name: CMD
      expand_in_command: true
      command: '(env ; echo "(%V)(%f)%f_%r_%d-%i%v.%x %f_%br_%bd_%g/") > test.log'
```

mdeweerd, 2025-05-20:
@set-soft That's perfect.

This is the equivalent of what I did - if "_PREFIX*_" is present then I get PREFIX*, if not the default:

```yaml
preflight:
  set_text_variables:
    - name: PRODUCT_CODE
      expand_in_command: true
      command: 'grep -oP "(?<=_)PREFIX[^_]*" <<< "%v" || echo "DEFAULT"'
```

closed 2025-05-20 by mdeweerd