INTI-CMNB/KiBot #826: [HOW] Getting text variables to expand inside a datasheet URL

CoolNamesAllTaken, 2025-05-21:
**What do you want to achieve?**
Describe the result you want to get with as much detail as possible.

I have some datasheets stored locally in a folder that is referred to by the environment variable ${KICAD_LIBS_DIR}/datasheets. When I run kibot in a docker container that is populated with the correct environment variables, I still get unresolved text variable errors from kibot due to text variables in symbol datasheet fields not being resolved before ERC.

**Do you have some PCB/Schematic to use as example?**
A small example can be attached, no need to attach a whole project. To attach files put them inside a .ZIP.
If you are working on a public repo a link to the project is ok.

I unfortunately can't share the full schematic, but can put together an example if it's helpful! I'm hopeful that I'm doing something very obvious and dumb and might be able to get to a resolution more easily.

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
kibot:
  version: 1

global:
  environment:
    # symbols: ${KICAD9_SYMBOL_DIR}
    # footprints: ${KICAD9_FOOTPRINT_DIR}
    # models_3d: ${KICAD9_3DMODEL_DIR}
    # templates: ${KICAD9_TEMPLATE_DIR}
    # user_templates: ${KICAD9_USER_DIR}/template
    extra_os:
      - name: KICAD_LIBS_DIR
        value: ${KICAD_LIBS_DIR}

preflight:
  erc: true
  drc: true
  fill_zones: true
  draw_fancy_stackup: false
  update_pcb_characteristics: true

outputs:

  ## Plots Outputs ##

  - name: "gerber"
    comment: "Gerber files for PCB manufacturing."
    type: gerber
    dir: +plots/gerber
    options:
      # generic layer options
      exclude_edge_layer: false
      exclude_pads_from_silkscreen: false
      plot_sheet_reference: true
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true
      line_width: 0.15

      # gerber options
      use_aux_axis_as_origin: false
      subtract_mask_from_silk: true
      use_protel_extensions: false
      gerber_precision: 4.5
      create_gerber_job_file: true
      use_gerber_x2_attributes: true
      use_gerber_net_attributes: false
    layers:
      - "copper"
      - "F.Mask"
      - "B.Mask"
      - "F.SilkS"
      - "B.SilkS"
      - "F.Fab"
      - "B.Fab"
      - "Edge.Cuts"
      - "User.Drawings"
      - "User.Comments"
  
  - name: "drill_with_map_pdf"
    comment: "Drill files for PCB manufacturing."
    type: excellon
    dir: +plots/drill
    options:
      # excellon options
      pth_and_npth_single_file: True
      map: "pdf"

  - name: "drill_map_dxf"
    comment: "Drill files for PCB manufacturing."
    type: excellon
    dir: +plots/drill
    options:
      # excellon options
      pth_and_npth_single_file: True
      map: "dxf"

  ## Fab Outputs ##

  - name: "schematic_pdf_annotated"
    comment: "Annotated PDF of the schematic."
    type: pdf_sch_print
    dir: +fab
    options:
      output: "%f-%i%I%v-annotated.%x"

  - name: "interactive_html_bom"
    comment: "Interactive HTML BOM."
    type: ibom
    dir: +fab
    options:
      include_tracks: true
      include_nets: true
      output: "%f-%i%I%v.%x" # %i=ibom
      blacklist_empty_val: true
      dark_mode: true
      highlight_pin1: true

  - name: "export_3d"
    comment: "3D model in STEP format."
    type: export_3d
    dir: +fab

  - name: "positions"
    comment: "Pick and place file."
    type: position
    dir: +fab
    options:
      format: ASCII
      only_smd: false

  - name: "positions-openpnp"
    comment: "Pick and place file with mirrored bottom components for OpenPNP."
    type: position
    dir: +openpnp
    options:
      format: ASCII
      only_smd: false
      include_virtual: true
      bottom_negative_x: true
  
  - name: "netlist_d356"
    comment: "Netlist in IPC-D-356 format."
    type: netlist
    dir: +fab
    options:
      format: "ipc"

  # - name: "diff"
  #   comment: "Diff file."
  #   type: diff
  #   dir: +fab

  ## BOM Outputs ##

  - name: "bom_csv"
    comment: "Bill of Materials in CSV format."
    type: bom
    dir: +bom
    options:
      csv:
        hide_pcb_info: true
        quote_all: true
      group_fields: [ "Manufacturer", "MPN" ]

      columns:
        - field: "References"
          name: "Reference"
        - field: "Value"
        - field: "Footprint"
        - field: "Datasheet"
        - field: "Quantity Per PCB"
          name: "Qty"
        - field: "Footprint Populate"
          name: "Populate"
        - field: "Standard Cost"
        - field: "Manufacturer"
        - field: "MPN"
        - field: "LCSC PN"
        - field: "Note"
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`)
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`)
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`)
- Using a desktop system? Which OS, distro and version?

Running KiBot 1.8.4 in [this](https://hub.docker.com/repository/docker/coolnamesalltaken/pantsforbirds-kibot/general) docker container based on an Ubuntu image.

**Additional context**
Add any other context about your setup/project.

I can run a shell inside the docker container and see that the environment variable for KICAD_LIBS_DIR is being passed in correctly, and resolves to a valid path to a folder inside the docker container. When I run my kibot export job, I get the following error:
```
ERROR:2 ERC errors detected (kibot.pre_erc - pre_erc.py:147)
ERROR:(unresolved_variable) Unresolved text variable
    @(1.016 mm, 0.6223 mm): Symbol U2 [BLB01]
    Sheet: / (kibot.pre_erc - pre_erc.py:162)
ERROR:(unresolved_variable) Unresolved text variable
    @(2.1336 mm, 0.6223 mm): Symbol U3 [BLB01]
    Sheet: / (kibot.pre_erc - pre_erc.py:162)
ERROR:ERC errors: 2 (kibot.gs - gs.py:899)
```
Looking into the schematic, the only unresolved text variable appears to be in the datasheet field.

[image]

set-soft, 2025-05-21:
> global:
>   environment:
>     # symbols: ${KICAD9_SYMBOL_DIR}
>     # footprints: ${KICAD9_FOOTPRINT_DIR}
>     # models_3d: ${KICAD9_3DMODEL_DIR}
>     # templates: ${KICAD9_TEMPLATE_DIR}
>     # [...]

What are you trying to do here? Doesn't look good. 
You are defining an environment variable called `KICAD_LIBS_DIR` that contains the text `${KICAD_LIBS_DIR}` 

> I can run a shell inside the docker container and see that the environment variable for KICAD_LIBS_DIR is being passed in correctly, and resolves to a valid path to a folder inside the docker container. When I run my kibot export job, I get the following error:

If that's the case then try removing the definition in the `environment` section, you say that the variable is already in the environment, no need to define it in KiBot.

I guess the confusion is about KiCad variables vs environment variables.
KiCad can expand environment variables in datasheet URLs, if the `KICAD_LIBS_DIR` environment variable is defined in your environment then it should work without any extra effort.

Your definition looks like a recursive one.

set-soft, 2025-05-21:
Note: in the title you mention "text variables", but you don't need them for this case.
Also: text variables (KiCad variables) can be defined in your project, again: no need to define them in KiBot.
KiBot can create KiCad variables that are dynamically adjusted, i.e. the hash for the last git commit, otherwise you just define them in your project.

CoolNamesAllTaken, 2025-05-21:
Awesome! I was completely overthinking it. Thank you very much for the guidance--when I don't attempt to override the environment variable, it gets pulled in by kibot just fine. I appreciate the prompt assistance, and thanks for all of your work on this awesome project!

closed 2025-05-21 by CoolNamesAllTaken