INTI-CMNB/KiBot #828: [BUG] DRC happily returns 0 when there are errors

oliv3r, 2025-05-23:
**Describe the bug**
The KiBot DRC handler will return 0 from the binary, when there are actually errors. For ERC, I do not see this issue, as ERC will fail actually ...

```yaml
kibot:
  version: 1

import:
  - file: 'kici_common.kibot.yaml'

preflight:
  drc:
    all_track_errors: true
    dir: 'reports'
    dont_stop: true
    enabled: true
    format: 'CSV,HTML,JSON,RPT'
    warnings_as_errors: true
  erc:
    dir: 'reports'
    dont_stop: true
    enabled: true
    format: 'CSV,HTML,JSON,RPT'
    warnings_as_errors: true
```

**To Reproduce**
```
% cd /tmp
% docker run \
    --env HOME='/tmp' \
    --interactive \
    --pull always \
    --rm \
    --tty \
    --user "$(id -u):$(id -g)" \
    --volume "$(pwd):/workdir" \
    --workdir "/workdir" \
    'ghcr.io/inti-cmnb/kicad9_auto_full:dev' \
    '/bin/bash'
$ git clone https://gitlab.com/ci-includes/kici.git
$ cd kici/hello
$ git reset --hard v0.10.0
$ export CI_JOB_NAME='KiCi'
$ kibot --defs-from-env  --plot-config '.kibot/kici_drc.kibot.yaml'
```

**Expected behavior**
When there are errors, we should always 'return -1' or similar so that CI/CD can catch these failed return values.

**Screenshots**
```
I have no name!@8d1bdcc025ab:/workdir/kici/hello$ kibot --defs-from-env --plot-config '.kibot/kici_drc.kibot.yaml'
Using SCH file: hello.kicad_sch
- Running the DRC
ERROR:1 DRC errors detected (kibot.pre_drc - pre_drc.py:151)
ERROR:(lib_footprint_mismatch) Footprint 'CP_Radial_D6.3mm_P2.50mm' does not match copy in library 'Capacitor_THT'.
    @(125.32 mm, 57.46 mm): Footprint C1
    Check: Violations (kibot.pre_drc - pre_drc.py:166)
ERROR:DRC errors: 1 (kibot.gs - gs.py:900)
Found 0 unique warning/s (1 total, 1 filtered)
```

**Additional context**
What I really want, in summary as I described a few times :p,
a) Warnings are Treated as Errors. While warnings are 'okay' in some cases, uses may want to fix warnings. Since pipelines cannot tell you if it's a warning or an error, all warnings would be blindly 'green' as nothing is deemed wrong.
b) When there are errors, the return code should be non-0, so pipelines can fail on this.
c) Kibot should be able to continue running in certain cases, to produce all outputs. E.g. when doing ERC/DRC, kibot should not stop on the first error, make the user fix it, and then 'oh hey, there was a nother error'. Instead all errors/warnings should be printed so we can see this output. (e.g. internally a global 'return_status = 0' and whenever something generates an error return value, `return_status = error` and in the end `exit return_status` (in poor mans pseudo code) :)

set-soft, 2025-05-23:
@oliv3r I don't see any bug here, if you ask to continue on error you are actually ignoring the errors. So I don't see any fail to comply with the specs, no bug.
Also note that KiBot doesn't return "-1" it returns specific codes indicating what failed. And note that we could be ignoring multiple errors, so which one will be reported?
I understand you want to stop, but this is something that can be *added*, this is why I label it as a new feature.

set-soft, 2025-05-23:
BTW: KiBot clearly differentiate warnings and error in its output.
You can parse its output to know about them.

set-soft, 2025-05-23:
The above patch adds an option that should exit with error for ignored errors 
Note that the behavior on ambiguous situations will be ... ambiguous.

closed 2025-05-23 by set-soft

oliv3r, 2025-05-23:
> [@oliv3r](https://github.com/oliv3r) I don't see any bug here, if you ask to continue on error you are actually ignoring the errors. So I don't see any fail to comply with the specs, no bug. Also [...]

Ahhah! I finally understand our miss-communication on this topic!!

You are absolutely right, all these things are not problems or bugs with KiBot!

However, this is not how traditional unix tools, or CI pipelines tend to work.

Lets take for example the following pipeline script:
```sh
ls /tmp/testfile
cp /tmp/testfile /tmp/testfile2
```
(Please ignore the fact that I use ls to check for the existance of a file, you are supposed to use `test` or `stat` even for that :p )

Now lets assume that `/tmp/testfile` does not exist. Doing this from the console, means `ls` fails with `no such file or directory`, and you would argue 'there is nothing wrong with the tool' so it should return 0.

So if `set -x` is done, which pipeline executors do, both commands would return a failure if they cannot list or copy the files, even though this is not a bug in the program.

Of course, parsing the output is certainly possible.

So in the end, especially when wanting to keep existing behavior, is introduce a flag, where during exit you'd have something:
```
if flag && error != 0; return error;
return status;
```

oliv3r, 2025-05-23:
BTW, we've added quite a few flags to influence this behavior over the past year. I wonder if it makes sense to revisit them and see what may be more logical. I'm just thinking out loud here, and can imagine that in hind-sight, '--treat-warnings-as-errors' and '--continue-on-errors' and '--return-error-as-failure' in the end is the summary what we've been trying to accomplish maybe?

set-soft, 2025-05-23:
> BTW, we've added quite a few flags to influence this behavior over the past year. I wonder if it makes sense to revisit them and see what may be more logical.

To do what after this?

oliv3r, 2025-05-23:
Don't know, just thinking out loud :)

set-soft, 2025-12-18:
Please post a small test case to reproduce the problem, the smallest KiBot config possible.
Explain, what do you expect and what do you get.
Don't use the whole KiCi as example.

oliv3r, 2026-01-04:
Hey @set-soft,

hope you had a good new-years!

I just checked, and did see a 'fix' but alas, with todays dev build the last case is still failling ...
```
Digest: sha256:13d163259c527beff1a6432d41d180e2cb9b35a20f606c0adf7f53f642bfbece
Status: Downloaded newer image for ghcr.io/inti-cmnb/kicad9_auto_full:dev
I have no name!@a025a1ef83a7:/workdir$ cd 'world'
I have no name!@a025a1ef83a7:/workdir/world$ . '.kibot/kici_conf.env'
I have no name!@a025a1ef83a7:/workdir/world$ for _k in $(grep "^KICI_.*=.*$" '.kibot/kici_conf.env'); do export "${_k%%'='*}"; done
I have no name!@a025a1ef83a7:/workdir/world$ export CI_JOB_NAME='KiCi'
I have no name!@a025a1ef83a7:/workdir/world$ kibot -e world.kicad_sch  --dont-stop --fail-on-ignored --defs-from-env --fail-on-warnings --plot-config '.kibot/kici_erc.kibot.yaml'; echo $?
- Running the ERC
Found 0 unique warning/s (2 total, 2 filtered)
39
I have no name!@a025a1ef83a7:/workdir/world$ kibot --version
KiBot 1.8.6 - Copyright 2018-2025, Salvador E. Tropea/INTI/John Beard - License: AGPL v3
I have no name!@a025a1ef83a7:/workdir/world$ kibot -e world.kicad_sch  --dont-stop --defs-from-env --fail-on-warnings --plot-config '.kibot/kici_erc.kibot.yaml'; echo $?
- Running the ERC
Found 0 unique warning/s (1 total, 1 filtered)
39
```

commit 558f52bd (2025-05-23): [CLI][Added] -F to return error even when using -D

- Aka --fail-on-ignored

Closes #828