INTI-CMNB/KiBot #832: [FEATURE]Specify resolution for diff

mdavidsaver, 2025-05-28:
I would like to produce higher resolution diffs, which I think involves passing `--resolution <dpi#>` to `kidiff`.  Maybe similar to how `--threshold` is handled?

https://github.com/INTI-CMNB/KiBot/blob/ad50432f27f5607a07505d27303fd162deb515d4/kibot/out_diff.py#L471-L472

mdavidsaver, 2025-05-28:
Is there a straightforward way for me to test `dev` with a github action build?

```yaml
    - uses: INTI-CMNB/KiBot@v2_dk9
```

fyi. [my case](https://github.com/BerkeleyLab/Marble/compare/main...mdavidsaver:Marble:kibot?expand=1#diff-08b83a0a4b70fef9e2c000749122f88b91a13b47c0cfc9d7379b8e84eea6fdeaR41)

set-soft, 2025-05-28:
This will pull the dev image. `v2_dk9` is `dev` for `KiCad 9`

mdavidsaver, 2025-05-28:
oops.  I had my GHA jobs mixed up.  `v2_dk9` has the desired effect.  Here is the default 150 vs. 300 DPI.

[image] [image]

set-soft, 2025-05-29:
Yes this means that the conversion from PDF to PNG failed, but we didn't get an error when running the external program.
I should add some message.
This is most probably because the converter run out of memory, you need four times the memory for 600 dpi, and the operations needs much more memory than the actual images.
Default KiCad page is A4, 210x297 mm = 8.27"x11.69", @ 600 dpi is 4962x7014 px = 33.2 Mi px even at 4 bytes per pixel this is just 132 MiB.
This should be handled without problems, but in practice the tools needs insane amounts of RAM for it. Don't know the exact reason.
And this is why 150 is the default. Using 600 is 16 times more RAM

closed 2025-05-29 by set-soft