INTI-CMNB/KiBot #839: [BUG] Fails when private elements are added on components

mroetsc, 2025-06-16:
**Describe the bug**
When defining a private element on a component, the generation crashes with the following error:
```sh
ERROR:While loading `/workspace/KICAD_FILE.kicad_sch` (kibot.gs - gs.py:897)
ERROR:text is not a string `Symbol('private')` (kibot.gs - gs.py:897)
```

**To Reproduce**
1. Go into Symbol Editor of any symbol (CTRL + E or via right click)
2. Add text and mark it as private in the Basic Properties section (see image)
3. Run anything that needs the schematic

This is how the `.kicad_sch` file will look:
```
(symbol "10k0_1_1"
  (text private "This is a private text field"
```

**Expected behavior**
As it is a private field, it should simply be ignored

**Screenshots**

[image]

**Environment (please complete the following information):**
Where are you running KiBot:
- With the `ghcr.io/inti-cmnb/kicad9_auto_full:latest` docker image

mroetsc, 2025-06-17:
Thanks for the fast reply, however, the issue persists with the `dev` image. The error message is the same.

I noticed, that you added the private field in the test like this:
https://github.com/INTI-CMNB/KiBot/blob/b12233bff1562aea6256f7d3878545f9a4bb2de8/tests/reference/9_0_1/test_v5.kicad_sch#L1850
which looks different than what i had:
```
(symbol "10k0_1_1"
  (text private "This is a private text field"
```

Does this make a difference? Why else could this still fail?

From the container:
```
$ kibot --version
KiBot 1.8.5 - Copyright 2018-2025, Salvador E. Tropea/INTI/John Beard - License: AGPL v3
```

set-soft, 2025-06-17:
Create a small file where it fails, it might be some other use of the "private" flag