INTI-CMNB/KiBot #840: [HOW] How to pass environment variable in KiBot action

Kampi, 2025-06-17:
**What do you want to achieve?**

I have a complex KiBot workflow, based on [this repository](https://github.com/nguyen-v/amulet_controller_kibot/tree/master), and I want to adjust the workflow to my project structure. My project structure contains an environment variable, called `KICAD_LIBRARY`, which contains the root path of my KiCad library (Symbols, Footprints, 3D, etc.).

I have the issue that the KiBot action in my current CI/CD workflow doesn´t use the correct variable.

Describe the result you want to get with as much detail as possible.

The KiBot action reports these warnings (see [this](https://github.com/Kampi/test/actions/runs/15698534040/job/44228311398,) run as an example):

```sh
WARNING:(W047) Missing 3D model for SW101: `/github/workspace/library/3D/Wuerth Elektronik/430182095816.stp` (kibot - out_base_3d.py:542)
WARNING:(W047) Missing 3D model for M101: `/github/workspace/library/3D/Ezurio/453-00001.stp` (kibot - out_base_3d.py:542)
WARNING:(W047) Missing 3D model for IC102: `/github/workspace/library/3D/Broadcom/APDS-9306-0[65](https://github.com/Kampi/test/actions/runs/15698534040/job/44228311398#step:13:66).stp` (kibot - out_base_3d.py:542)
WARNING:(W047) Missing 3D model for IC103: `/github/workspace/library/3D/Bosch/BME680.stp` (kibot - out_base_3d.py:542)
WARNING:(W047) Missing 3D model for BT101: `/github/workspace/library/3D/Battery/CR2032.stp` (kibot - out_base_3d.py:542)
WARNING:(W047) Missing 3D model for BT101: `/github/workspace/library/3D/Keystone/3002.stp` (kibot - out_base_3d.py:542)
```

I define the variable via the `env` section of my pipeline:

```sh
  KICAD_LIBRARY: /github/workspace/library
````

And the variable is passed into the action

```sh
Run INTI-CMNB/KiBot@v2_dk9
/usr/bin/docker run --name e554bb36a45e5afc81419a139bfe2_f788f7 --label 717715 --workdir /github/workspace --rm -e "kicad_board" -e "kibot_config" -e "kibot_variant" -e "MOUSER_KEY" -e "KICAD_LIBRARY" 
```

**Do you have some PCB/Schematic to use as example?**

My test playground for migrating the templates from Vincent:

https://github.com/Kampi/test

**Do you have some configuration file (.kibot.yaml) that you are using?**

See the project

**Environment (please complete the following information):**

- GitHub CI/CD

**Additional context**

I´m not sure if it´s possible to test it locally but I try to figure it out later.

Kampi, 2025-06-17:
Hi @set-soft,

you can find the PCB [here](https://github.com/Kampi/test/tree/Add-KiCad-Templates/hardware). The setup is the following:

You download the library from [here](https://github.com/Kampi/KiCad) and then you define an environment variable called `KICAD_LIBRARY` with the path to the library root on your system. All footprints, symbols and 3D models are linked to this variable (see the screenshot as example).

[image]

This setup should be used in the CI/CD pipeline. My [old workflow](https://github.com/Kampi/KiCad/blob/master/.github/workflows/production.yml) uses this variable without issues, but it´s based on the KiBot Docker container and does not use the KiBot action in the workflow.

Now the question is:
How can I use the library repository in this new workflow correctly so the paths to the 3D models are resolved correctly? So I must define the variable `KICAD_LIBRARY` in the workflow and pass it into the KiBot action with the correct path.

Does this explanation help you to understand the issue, or do you need additional information?

set-soft, 2025-06-18:
KiBot is getting your environment variable, you can see it here:

```
WARNING:(W047) Missing 3D model for M101: `/github/workspace/library/3D/Ezurio/453-00001.stp` (kibot - out_base_3d.py:542)
```

It doesn't say `${KICAD_LIBRARY}/3D/Ezurio/453-00001.stp` or similar.

You are downloading the libs here:
```
      - name: Dependencies
        run: |
          git clone https://github.com/Kampi/KiCad.git library
          echo ${KICAD_LIBRARY}
```

So they will be in something like `/github/workspace/library`, assuming `/github/workspace` is the current working directory, something you can verify running the `pwd` command, you should also add `ls -la` after the clone

But then the GitHub action is running using the following mount point:  `-v "/home/runner/work/test/test":"/github/workspace"`  
Which I think isn't the same as `/github/workspace/` from the host.

There are various solutions to this, the simplest is avoid messing with manual things, just make `https://github.com/Kampi/KiCad.git` a submodule of your main repo and configure the checkout to include submodules, i.e.

```
    - uses: actions/checkout@v4
      with:
        submodules: recursive
```

You won't need to do an extra clone in this way. This should work.

Another approach is to just don't use GitHub actions to run KiBot, this is the simplest way to run KiBot, but also the less flexible. Not recommended for advanced use.