INTI-CMNB/KiBot #859: [BUG] Output BOM takes forever - high CPU and memory usage

cyberreefguru, 2025-08-27:
I've been using Kibot for a while now and recently started getting out of memory errors from Github when generating releases. After a few days of investigating, I isolated the problem to two main things:

1 - using the compress output seems to storage everything in memory until it is complete. My VM was using over 18GB of memory before writing out the zip file, which was about 18MB in size. I've worked around that by not using the compress output and just zipping after the fact.

2 - in the process of debugging the first issue, I noticed my panel was taking a really long time to complete. Much debugging later, I noticed generating the position file was taking ~7 mins. When run manually via the application it takes less than a second.

**To Reproduce**
kibot -c ../../../kicad-extras/kibot/crgm.kibot.yaml -e ../aquarius-client-hw.kicad_sch -b pcb/aquarius-client-hw-panel.kicad_pcb -d production create_pnp_jlc

where create_pnp_jlc is:

```
- name: 'create_pnp_jlc'
    comment: 'Generate JLC PnP File'
    run_by_default: false
    type: position
    dir: pnp
    options:
      output: '%f-jlc-cpl.%x'
      format: CSV
      units: millimeters
      separate_files_for_front_and_back: false
      only_smd: false
      exclude_filter: 'exclude_common'
      variant: jlcpcb
      columns:
        - id: Ref
          name: Designator
        - id: PosX
          name: "Mid X"
        - id: PosY
          name: "Mid Y"
        - id: Side
          name: Layer
        - id: Rot
          name: Rotation
```

and

```
filters:
  - name: 'exclude_common'
    type: 'generic'
    comment: 'Excludes test points, logos, and fiducials'
    exclude_any:
      - column: Reference
        regex: '^TP[0-9]*'
      - column: Reference
        regex: '^FID'
      - column: Reference
        regex: '^LOGO'
      - column: Reference
        regex: '^KiKit_'
      - column: Reference
        regex: '^kibuzzard-'
  - name: only_jlc_parts
    comment: 'Only parts with JLC (LCSC) code'
    type: generic
    include_only:
      - column: 'LCSC'
        regex: '^C\d+'
  - name: fix_rotation
    comment: 'Adjust rotation for JLC'
    type: rot_footprint
    rotations:
      - ["^QFN-28-1EP_6x6mm_P0.65mm_EP4.25x4.25mm", -90.0]
      - ["^JST_SH_BM08B-SRSS-TB_1x08-1MP_P1.00mm_Vertical", 180.0]
      - ["^JST_SH_BM06B-SRSS-TB_1x06-1MP_P1.00mm_Vertical", 180.0]
      - ["^JST_SH_BM04B-SRSS-TB_1x04-1MP_P1.00mm_Vertical", 180.0]
      - ["^JST_SH_BM03B-SRSS-TB_1x03-1MP_P1.00mm_Vertical", 180.0]
      - ["^LED_WS2812B-2020_PLCC4_2.0x2.0mm", 0.0]
      - ["^SOT-523", 180.0]
      - ["^SOT-666", -90.0]
      - ["^XF2M-1415-1A", 180.0]
      # - ["^TI_SO-", 270.0]
      # - ["^SO-", 270.0]
      # - ["^HTSSOP-", 270.0]
      # - ["^SSOP-", 270.0]
      # - ["^MSOP-", 270.0]
      # - ["^USB_C_Receptacle_Palconn", 180.0]
      - ["^SOTFL50P160X60", 270]
      - ["^CP_Elec_6.3x7.7", 0]
      - ["^Bourns_TC33X_Vertical", -90]
    offsets:
      - ["^TO-252-2", "3.3,0"]
      - ["^Converter_ACDC_MeanWell_IRM-10-xx_THT", "-19.2, 5.5"]
      - ["^GCT_USB4105-GF-A", "0, -2.5"]
      - ["^ESP32-S3-WROOM-1-N8R2", "0, 3.65"]
      - ["^XF2C_1415-1A", "0, -0.37"]

variants:
  - name: default
    comment: 'Just a place holder for the rotation filter'
    type: kibom
    variant: default
  - name: jlcpcb
    comment: 'JLCPCB requires some rotational transforms'
    type: kibom
    variant: jlcpcb
    pre_transform: fix_rotation
```

All available on my GitHub page: https://github.com/crgmakes/aquarius-client-hw and https://github.com/crgmakes/kicad-extras

**Expected behavior**
Generating the BOM should not take 7 mins. Running manually via command line takes about 7 seconds:

```
root@e32c6cff12de:/home/working/aquarius-client-hw/pcb/panel# date
Wed Aug 27 22:20:38 UTC 2025
root@e32c6cff12de:/home/working/aquarius-client-hw/pcb/panel# kicad-cli pcb export pos ./pcb/aquarius-client-hw-panel.kicad_pcb 
Wrote position data to 'aquarius-client-hw-panel.pos'.
root@e32c6cff12de:/home/working/aquarius-client-hw/pcb/panel# date
Wed Aug 27 22:20:45 UTC 2025
root@e32c6cff12de:/home/working/aquarius-client-hw/pcb/panel# 
```

**Screenshots**

Screen shots of docker using +4GB of memory and taking ~7mins to complete. 

[image]
[image]

**Environment (please complete the following information):**
Where are you running KiBot:
- Started as CI/CD in git then moved to docker container running on my computer for more detailed debugging.
- I'm using ghcr.io/inti-cmnb/kicad9_auto_full:latest
- Local is MacOS with Docker Desktop 4.44.3 (202357)

**Additional context**
My panel is attached. All scripts and kibot.yaml are on git for reference.

[aquarius-client-hw-panel.kicad_pcb.zip](https://github.com/user-attachments/files/22014466/aquarius-client-hw-panel.kicad_pcb.zip)

I looked at the source code and nothing jumps out but Im really not a python expert by any stretch.

Trying to understand if it's me or kibot or something in between.

Thanks.

cyberreefguru, 2025-08-28:
> Which version are you using? Did you try the _dev_ branch? See [#842](https://github.com/INTI-CMNB/KiBot/issues/842)

I am using whatever version is installed in the Docker image -- looks like KiBot 1.8.4. I did check the open tickets for any reported issues but didn't look at the closed tickets.

#842 is a memory issue and hints at a potential performance hit - I worked around my memory problem but at least locally it still taking an excessive amount of time compared to the local CLI invocation.

I will attempt to get the dev branch installed and see what happens.

-Tom

set-soft, 2025-08-28:
> I am using whatever version is installed in the Docker image

You also have docker images for the *dev* branch.

cyberreefguru, 2025-10-02:
I never did - I stopped using the compress target and did other optimizations which reduced memory and the time it was taking. On a larger board that might not work, though.  I will close the issue. Thanks!

closed 2025-10-02 by cyberreefguru